pix/heystack-nrf5x #1: make[1]: hex2bin.py: No such file or directory

kiam001, 2024-08-26:
cant compile

kiam001, 2024-08-26:
pip install IntelHex fix issue

closed 2024-08-26 by kiam001

5Breeze, 2025-02-20:
I still can't use it after installing it.
`root@HomeService:/home/FindMy/heystack-nrf5x-master# make all
dirname: missing operand
Try 'dirname --help' for more information.
make -C nrf52810/armgcc GNU_INSTALL_ROOT=/home/FindMy/heystack-nrf5x-master/nrf-sdk/gcc-arm-none-eabi-6-2017-q2-update/bin/ MAX_KEYS=500 HAS_DEBUG=0 HAS_BATTERY=0 KEY_ROTATION_INTERVAL=3600 ADVERTISING_INTERVAL=1000 RANDOM_ROTATE_KEYS=1 nrf52810_xxaa-dcdc bin_nrf52810_xxaa-dcdc
make[1]: Entering directory '/home/FindMy/heystack-nrf5x-master/nrf52810/armgcc'
DONE nrf52810_xxaa-dcdc
merging application and softdevice... : nrf52810_xxaa-dcdc
mergehex -m _build/nrf52810_xxaa-dcdc.hex ../../nrf-sdk/nRF5_SDK_15.3.0_59ac345/components/softdevice/s112/hex/s112_nrf52_6.1.1_softdevice.hex -o _build/nrf52810_xxaa-dcdc_s112.hex
Parsing input files.
Merging file "nrf52810_xxaa-dcdc.hex" into output.
Merging file "s112_nrf52_6.1.1_softdevice.hex" into output.
Storing merged file.
hex2bin.py _build/nrf52810_xxaa-dcdc_s112.hex _build/nrf52810_xxaa-dcdc_s112.bin
make[1]: hex2bin.py: No such file or directory
make[1]: *** [../../Makefile.common:84: bin_nrf52810_xxaa-dcdc] Error 127
make[1]: Leaving directory '/home/FindMy/heystack-nrf5x-master/nrf52810/armgcc'
make: *** [Makefile:63: nrf52810_xxaa-dcdc] Error 2
root@HomeService:/home/FindMy/heystack-nrf5x-master# pip install IntelHex
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
root@HomeService:/home/FindMy/heystack-nrf5x-master# sudo apt install python3-intelhex
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-intelhex is already the newest version (2.3.0-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@HomeService:/home/FindMy/heystack-nrf5x-master# make all
dirname: missing operand
Try 'dirname --help' for more information.
make -C nrf52810/armgcc GNU_INSTALL_ROOT=/home/FindMy/heystack-nrf5x-master/nrf-sdk/gcc-arm-none-eabi-6-2017-q2-update/bin/ MAX_KEYS=500 HAS_DEBUG=0 HAS_BATTERY=0 KEY_ROTATION_INTERVAL=3600 ADVERTISING_INTERVAL=1000 RANDOM_ROTATE_KEYS=1 nrf52810_xxaa-dcdc bin_nrf52810_xxaa-dcdc
make[1]: Entering directory '/home/FindMy/heystack-nrf5x-master/nrf52810/armgcc'
DONE nrf52810_xxaa-dcdc
merging application and softdevice... : nrf52810_xxaa-dcdc
mergehex -m _build/nrf52810_xxaa-dcdc.hex ../../nrf-sdk/nRF5_SDK_15.3.0_59ac345/components/softdevice/s112/hex/s112_nrf52_6.1.1_softdevice.hex -o _build/nrf52810_xxaa-dcdc_s112.hex
Parsing input files.
Merging file "nrf52810_xxaa-dcdc.hex" into output.
Merging file "s112_nrf52_6.1.1_softdevice.hex" into output.
Storing merged file.
hex2bin.py _build/nrf52810_xxaa-dcdc_s112.hex _build/nrf52810_xxaa-dcdc_s112.bin
make[1]: hex2bin.py: No such file or directory
make[1]: *** [../../Makefile.common:84: bin_nrf52810_xxaa-dcdc] Error 127
make[1]: Leaving directory '/home/FindMy/heystack-nrf5x-master/nrf52810/armgcc'
make: *** [Makefile:63: nrf52810_xxaa-dcdc] Error 2
root@HomeService:/home/FindMy/heystack-nrf5x-master# pip install IntelHex
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
^[[Aroot@HomeService:/home/FindMy/heystack-nrf5x-master# sudo apt install python3-intelhex
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-intelhex is already the newest version (2.3.0-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.`

5Breeze, 2025-02-20:
Finally, I used virtual env to fix it.