pix/heystack-nrf5x #3: Key Rotation not working when interval is 600 seconds

ctandi, 2024-09-07:
Hi,

first of all, thank you for this awesome firmware and your work!

However, when compiling it with a key rotation interval of 600 seconds, the firmware seems to get stuck with the initially selected advertising key.

That's how I compiled it
`make stflash-patched ADV_KEYS_FILE=~/Dokumente/Coding/heystack-nrf5x/tools/output/BEC26Q_keyfile GNU_INSTALL_ROOT=~/Dokumente/Coding/heystack-nrf5x/nrf-sdk/gcc-arm-none-eabi-6-2017-q2-update/bin/ KEY_ROTATION_INTERVAL=600 MAX_KEYS=60 HAS_DEBUG=1`

and this is the output

```
<info> app: last_filled_index: 58
<info> app: Starting advertising
<info> app: ble_set_mac_address: D9:B8:D3:75:9D:01
<info> app: ble_set_max_tx_power: 8 dBm failed
<info> app: ble_set_max_tx_power: 7 dBm failed
<info> app: ble_set_max_tx_power: 6 dBm failed
<info> app: ble_set_max_tx_power: 5 dBm failed
<info> app: ble_set_max_tx_power: 4 dBm
<info> app: Rotating key: 7
```
This happens to me on both the nRF52810 and the nRF52832. With the nRF52832, I waited over an hour with no visible changes.

It seems to work when I set it to 30 seconds:

```
<info> app: last_filled_index: 58
<info> app: Starting advertising
<info> app: ble_set_mac_address: E1:3B:86:EC:1E:84
<info> app: ble_set_max_tx_power: 8 dBm failed
<info> app: ble_set_max_tx_power: 7 dBm failed
<info> app: ble_set_max_tx_power: 6 dBm failed
<info> app: ble_set_max_tx_power: 5 dBm failed
<info> app: ble_set_max_tx_power: 4 dBm
<info> app: Rotating key: 37
Debug: 38 22792 strtt.cpp:326 readRtt(): Chanel: 0 readed: 322
<info> app: ble_set_mac_address: C0:11:4B:DD:42:A5
<info> app: ble_set_max_tx_power: 8 dBm failed
<info> app: ble_set_max_tx_power: 7 dBm failed
<info> app: ble_set_max_tx_power: 6 dBm failed
<info> app: ble_set_max_tx_power: 5 dBm failed
<info> app: ble_set_max_tx_power: 4 dBm
<info> app: Rotating key: 49
Debug: 39 52794 strtt.cpp:326 readRtt(): Chanel: 0 readed: 53
<info> app: ble_set_mac_address: D1:9E:EF:DE:57:EF
Debug: 40 52798 strtt.cpp:326 readRtt(): Chanel: 0 readed: 269
<info> app: ble_set_max_tx_power: 8 dBm failed
<info> app: ble_set_max_tx_power: 7 dBm failed
<info> app: ble_set_max_tx_power: 6 dBm failed
<info> app: ble_set_max_tx_power: 5 dBm failed
<info> app: ble_set_max_tx_power: 4 dBm
<info> app: Rotating key: 50
```

In addition to the output, I can watch the MAC changing on a BLE scanner app.

I haven't tried longer or shorter intervals apart from 600 and 30 seconds, but since 600 is probably a relatively common value, I thought I might as well report it.

ctandi, 2024-09-09:
I think I have identified the issue: `APP_TIMER_CONFIG_RTC_FREQUENCY` in `sdk_config.h` (or `APP_TIMER_PRESCALER` in `ble_stack.h` for older SDK versions) is set to zero, allowing for timers of up to 511 seconds (according to [this document](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52832.ps.v1.1%2Frtc.html)). So in the current configuration, it just overflows every time.

I'm going to try setting the prescaler to 255 later today and see if that helps.

pix, 2024-09-09:
Setting it to 31 might be more appropriate ? (4.6 hours) before rollover. I don't know why it seems to work with the nrf51 with the default prescaler (maybe it doesn't ? I don't remember the effective fw version I deployed)

I'm waiting for an order for more devices, as experimenting with the already used ones is not practical.

pix, 2024-09-09:
At least, a prescaler value of 31 is working on NRF51.

ctandi, 2024-09-10:
> Setting it to 31 (the max in the doc)

I set it to 255 yesterday and can confirm that this is working on the nRF52810.

> I don't know why it seems to work with the nrf51 with the default prescaler (maybe it doesn't ? I don't remember the effective fw version I deployed)

Based on the documentation that I've read so far, I would assume that it doesn't :D

pix, 2024-09-17:
I pushed an update. The prescaler is set to 31 on all devices. This will allow a max rotation time of more than 4h30. A static assert ensure the timer is within bounds.

PuNS3c, 2024-09-23:
> At least, a prescaler value of 31 is working on NRF51:
> 

Not that related to the topic, but I am interested - What is this setup you have for flashing?
Looking for making something like this myself..

pix, 2024-09-23:
You mean this: https://www.printables.com/model/419067-pcb-workstation-kit-v1-with-needle-probes ?

A colleague printed it for me, so I don"t have the exact STL model. But it should be a remix of this one.

closed 2024-10-13 by pix