pix/heystack-nrf5x #4: Flashing NRF52832Y

jrperson, 2024-10-07:
I am having an issue trying to flash the NRF52832Y board and was wondering if anyone has configurations for successfully doing so.

I tried flashing it using an ST-Link V2 with the default Openocd.cfg
```
source [find interface/stlink-v2.cfg]
source [find target/nrf52.cfg]
```

This results in an error
> nRF52 device has a CTRL-AP dedicated to recover the device from AP lock.
> A high level adapter (like a ST-Link) you are currently using cannot access
> the CTRL-AP so 'nrf52_recover' command will not work.
> Do not enable UICR APPROTECT.

I then tried flashing it using a Raspberry Pi
```
source [find interface/raspberrypi-native.cfg]
transport select swd
source [find target/nrf52.cfg]
```

This get me an error as well
> Error: Error connecting DP: cannot read IDR

I also tried "transport select hla_swd" but get error
> Debug adapter doesn't support 'hla_swd' transport

pix, 2024-10-10:
Yup. Some NRF52 are locked. You either need a real stlink (not a cheap clone) or another programmer (a jlink, a genuine stlink, a black magic probe, a daplink...)

Alternatively, you can flash Black Magic Probe firmware on you stlink:
- Get the black magic probe source firmware
- Modify the stlink.ini file:

```
[binaries]
c = 'arm-none-eabi-gcc'
cpp = 'arm-none-eabi-g++'
ld = 'arm-none-eabi-gcc'
ar = 'arm-none-eabi-ar'
nm = 'arm-none-eabi-nm'
strip = 'arm-none-eabi-strip'
objcopy = 'arm-none-eabi-objcopy'
objdump = 'arm-none-eabi-objdump'
size = 'arm-none-eabi-size'

[host_machine]
system = 'bare-metal'
cpu_family = 'arm'
cpu = 'arm'
endian = 'little'

[project options]
probe = 'stlink'
targets = 'cortexm,nrf,nxp,rp,sam,stm'
rtt_support = true
stlink_swim_nrst_as_uart = false
bmd_bootloader = true
```

- Flash the resulting firmware on your stlink:

```
  14045   st-flash write build/blackmagic_stlink_bootloader.bin 0x8000000                                                                                                                                                                     
  14046   st-flash write build/blackmagic_stlink_firmware.bin 0x8002000  
```

pix, 2024-10-13:
Have you manage to flash it ?

jrperson, 2024-10-14:
I haven't.  I have been struggling with building BMP for the stlink.  In the current version when I make and compile I get the firmware.bin, but I do not get the bootloader.bin.  I get a bootloader.elf.p file when I do the make.

I ordered a BMP v2.3 board so I can just use it without having to fiddle with things.  I'll update when I get the board and try to flash the nrf52832

smierd, 2025-03-06:
Hey Pix, if you could send me a firmware file to flash the ST-Link V2 that would be appreciated. Tried building with blackmagic but getting some errors that I can't seem to find a fix for. 

Not related to this project but the error is:
```
Cloning into 'libopencm3'...
warning: Could not find remote branch head to clone.
fatal: Remote branch head not found in upstream origin

src/platforms/common/stm32/meson.build:100:0: ERROR: Git command failed: ['/usr/bin/git', '-c', 'advice.detachedHead=false', 'clone', '--depth', '1', '--branch', 'head', 'https://github.com/blackmagic-debug/libopencm3', 'libopencm3']
```

libopencm3 is a valid library but seems to branch structure has changed and I can't figure out how to change Meson to pick the master branch.

closed 2025-05-13 by jrperson