pix/heystack-nrf5x #7: 50_NRF_keyfile Structure Request

CloudAllowed, 2024-10-12:
Hey Folks!

Can anybody support me with an example of the 50_NRF_keyfile.
I am struggling with the correct format.

BR & many THX Matthis

CloudAllowed, 2024-10-12:
My current has only one line/key:
`h4xsf3xXXXqXeXXpXX2/Sx4+oci93KgXsXXnxX==`

And I get this error message:

> The key was not patched correctly!

> merging application and softdevice... : nrf51822_xxac
mergehex -m _build/nrf51822_xxac.hex /home/matthis/nrf5x/nRF5_SDK_12.3.0_d7731ad/components/softdevice/s130/hex/s130_nrf51_2.0.1_softdevice.hex -o _build/nrf51822_xxac_s130.hex
Parsing input files.
Merging file "nrf51822_xxac.hex" into output.
Merging file "s130_nrf51_2.0.1_softdevice.hex" into output.
Storing merged file.
/home/matthis/.local/share/pipx/venvs/nrfutil/bin/hex2bin.py _build/nrf51822_xxac_s130.hex _build/nrf51822_xxac_s130.bin
Patching nrf51822_xxac
cp _build/nrf51822_xxac_s130.bin _build/nrf51822_xxac_s130_patched.bin
xxd -p -c 100000 /home/matthis/nrf5x/nrf51822/armgcc/1_NRF_Key | xxd -r -p | dd of=_build/nrf51822_xxac_s130_patched.bin skip=1 bs=1 seek=123292 conv=notrunc
81+0 records in
81+0 records out
81 bytes copied, 0.0010693 s, 75.8 kB/s
xxd -p -c 100000 _build/nrf51822_xxac_s130_patched.bin | grep -q 454e444f464b455953454e444f464b455953454e444f464b45595321 || (echo "The key was not patched correctly!"; exit 1)
"/home/matthis/nrf5x/gcc-arm-none-eabi-10.3-2021.10/bin/arm-none-eabi-objcopy" -I binary -O elf32-littlearm -B arm _build/nrf51822_xxac_s130_patched.bin _build/nrf51822_xxac_s130_patched.elf
flashing application and softdevice...␍
openocd -f openocd.cfg -c "init; halt; nrf51 mass_erase; program _build/nrf51822_xxac_s130_patched.bin verify; reset; exit";
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.305418
Info : [nrf51822.cpu] Cortex-M0 r0p0 processor detected
Info : [nrf51822.cpu] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for nrf51822.cpu on 3333
Info : Listening on port 3333 for gdb connections
[nrf51822.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x61000000 pc: 0x00011434 msp: 0x20003fc8
Info : nRF51822-QFAA(build code: H0) 256kB Flash, 16kB RAM
Info : Mass erase completed.
Info : A reset or power cycle is required if the flash was protected before.
[nrf51822.cpu] halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
Warn : Adding extra erase range, 0x0001ea5c .. 0x0001ebff
** Programming Finished **
** Verify Started **
** Verified OK **

pix, 2024-10-13:
You need to use the `generate_keys.py` script in `tools`.

```
python generate_keys.py -n 50 -p 50_NRF
```

CloudAllowed, 2024-10-14:
Same issue with the key generated in this way. Same error messages. 

Parhaps an issue with old or wrong libraries. I will setup the environment again blank from the scretch.
Will share my results here. THX.

> You need to use the `generate_keys.py` script in `tools`.
> 
> ```
> python generate_keys.py -n 50 -p 50_NRF
> ```
> 
> File structure is:
> 
> ```
> [8 bit key count (ignored)][key0 in binary][...][keyN in binary]
> ```

pix, 2024-10-15:
```
81+0 records in
81+0 records out
```

These should be multiple of 28.
Are you sure you picked the file ending in `_keyfile` ?

Alternatively, you could try the `nrf-patch-log.py` in `tools`

CloudAllowed, 2024-10-15:
I'm on vacation the next couple of days.
I'll give it a try after my return.
Many thanks!

CloudAllowed, 2024-11-14:
I rolled out the system with all dependencies from the scretch and generated a key with the tools/generate_keys.py but my error message is the same:

> 81+0 records in
> 81+0 records out
> 81 bytes copied, 0.0010693 s, 75.8 kB/s
> xxd -p -c 100000 _build/nrf51822_xxac_s130_patched.bin | grep -q 454e444f464b455953454e444f464b455953454e444f464b45595321 || [...]

Anybody an idea?

pix, 2024-11-15:
Can you show the make command ? This will work:

```
cd tools
python generate_keys.py -n 50 -p SmallTag1
cd ../nrf51822/armgcc
make patched_nrf51822_xxac GNU_INSTALL_ROOT=/home/pix/work/openhaystack/heystack-nrf5x/nrf51822/armgcc/../../nrf-sdk/arm-gnu-toolchain-13.3.rel1-x86_64-arm-none-eabi/ ADV_KEYS_FILE=../../tools/output-SmallTag1/SmallTag1_keyfile
```

From what I'm seeing:
1/ There's a error with your keyfile, it should not copy 81 bytes. The format is wrong.
2/ If make doesn't end with `make: *** [../../Makefile.common:101: patched_nrf51822_xxac] Error 1` it means it worked.

Here's the log of a failed patch:

```
2384+0 records in
2384+0 records out
2384 bytes (2.4 kB, 2.3 KiB) copied, 0.00698527 s, 341 kB/s
xxd -p -c 100000 _build/nrf51822_xxac_s130_patched.bin | grep -q 454e444f464b455953454e444f464b455953454e444f464b45595321 || (echo "The key was not patched correctly!"; exit 1)
The key was not patched correctly!
make: *** [../../Makefile.common:101: patched_nrf51822_xxac] Error 1
```

And here's a valid one:

```
xxd -p -c 100000 ../../tools/output-SmallTag1/SmallTag1_keyfile | xxd -r -p | dd of=_build/nrf51822_xxac_s130_patched.bin skip=1 bs=1 seek=116864 conv=notrunc
1400+0 records in
1400+0 records out
1400 bytes (1.4 kB, 1.4 KiB) copied, 0.00504892 s, 277 kB/s
xxd -p -c 100000 _build/nrf51822_xxac_s130_patched.bin | grep -q 454e444f464b455953454e444f464b455953454e444f464b45595321 || (echo "The key was not patched correctly!"; exit 1)
"/home/pix/work/openhaystack/heystack-nrf5x/nrf51822/armgcc/../../nrf-sdk/arm-gnu-toolchain-13.3.rel1-x86_64-arm-none-eabi//bin/arm-none-eabi-objcopy" -I binary -O elf32-littlearm -B arm _build/nrf51822_xxac_s130_patched.bin _build/nrf51822_xxac_s130_patched.elf
```

You see, your patch worked (even though the keyfile you used isn't)

CloudAllowed, 2024-11-16:
> Can you show the make command ? 

Generating keyfile with one key:

`sudo python3 ./generate_keys.py  -n 1 -p 1-record
Using python3
Output will be written to output-1-record/`

Flashing device:

`sudo make stflash-nrf51822_xxac-patched 'ADV_KEYS_FILE=../../tools/output-1-record/1-record_keyfile'
merging application and softdevice... : nrf51822_xxac
mergehex -m _build/nrf51822_xxac.hex ../../nrf-sdk/nRF5_SDK_12.2.0/components/softdevice/s130/hex/s130_nrf51_2.0.1_softdevice.hex -o _build/nrf51822_xxac_s130.hex
Parsing input files.
Merging file "nrf51822_xxac.hex" into output.
Merging file "s130_nrf51_2.0.1_softdevice.hex" into output.
Storing merged file.
/usr/share/python3-intelhex/hex2bin.py _build/nrf51822_xxac_s130.hex _build/nrf51822_xxac_s130.bin
Patching nrf51822_xxac
cp _build/nrf51822_xxac_s130.bin _build/nrf51822_xxac_s130_patched.bin
xxd -p -c 100000 ../../tools/output-1-record/1-record_keyfile | xxd -r -p | dd of=_build/nrf51822_xxac_s130_patched.bin skip=1 bs=1 seek=117656 conv=notrunc
28+0 records in
28+0 records out
28 bytes copied, 0.00107226 s, 26.1 kB/s
xxd -p -c 100000 _build/nrf51822_xxac_s130_patched.bin | grep -q 454e444f464b455953454e444f464b455953454e444f464b45595321 || (echo "The key was not patched correctly!"; exit 1)
"/usr/bin/arm-none-eabi-objcopy" -I binary -O elf32-littlearm -B arm _build/nrf51822_xxac_s130_patched.bin _build/nrf51822_xxac_s130_patched.elf
flashing application and softdevice...␍
openocd -f openocd.cfg -c "init; halt; nrf51 mass_erase; program _build/nrf51822_xxac_s130_patched.bin verify; reset; exit";
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.305418
Info : [nrf51.cpu] Cortex-M0 r0p0 processor detected
Info : [nrf51.cpu] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for nrf51.cpu on 3333
Info : Listening on port 3333 for gdb connections
[nrf51.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x61000000 pc: 0x00011434 msp: 0x20003fd8
Info : nRF51822-QFAA(build code: CA/C0) 256kB Flash, 16kB RAM
Info : Mass erase completed.
Info : A reset or power cycle is required if the flash was protected before.
[nrf51.cpu] halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
Warn : Adding extra erase range, 0x0001d184 .. 0x0001d3ff
** Programming Finished **
** Verify Started **
** Verified OK **`

Record seems correct:

`28+0 records in
28+0 records out
28 bytes copied, 0.00107226 s, 26.1 kB/s`

But does not work.

pix, 2024-11-19:
You can build a debug enabled firmware using

```
make clean
make stflash-nrf51822_xxac-patched ADV_KEYS_FILE=../../tools/output-1-record/1-record_keyfile HAS_DEBUG=1
```

Once flashed you can use the `strtt` tool to see the debug messages.

Do you see beacons using your phone and a BLE scanner ?