pix/heystack-nrf5x #8: Support for nRF52805?

ctandi, 2024-10-26:
I just discovered this super cheap set by HolyIoT: https://www.aliexpress.com/item/1005007083911385.html

@pix do you think this device/soc can be supported? The nRF52805 seems to have almost the same spec as the nRF52810. I just have very little experience developing firmwares for nRF chips, so I thought I might as well ask before ordering.

Thanks!

pix, 2024-10-29:
Hi o/

According to [this](https://www.nordicsemi.com/Products/nRF52805/Download), It needs a SDK 17. It was on my todolist to support newer SDK :)

I just tried to swap sdk-15.3 (current used version) and sdk-17 and it compiles for nRF52810. So I guess there isn't much work left to do. I'm a bit busy right now, but until it arrive in your mailbox, I should be able to push support for it :P

ctandi, 2024-10-29:
Awesome, I've just ordered a pair :D Thanks!

pix, 2024-10-30:
I managed to compile a binary for nRF52805  (don't know if it'll work). 

Current SDKs are missing some files/functions for nRF52805. Using https://github.com/NordicSemiconductor/nrfx/ to replace the nrfx module in the SDK allows to build the project.

pix, 2024-10-31:
I pushed a branch with the updated SDK and bunch of other stuff:

https://github.com/pix/heystack-nrf5x/tree/nrf52805-support

SDK download and patching is required. Remove the content of `nrf-sdk` and use `make sdk` to download the updated files.

ctandi, 2024-10-31:
Awesome, thanks! The parcel is still a couple days away, but once it's there I'll try it out.