pix/heystack-nrf5x #9: Problem flashing nrf52832 (holyiot yj-16013)

tobox, 2024-11-13:
The whole compile process seems to be OK. But openocd won't flash the chip using my ST-Link V2 flasher:

When I run:

`openocd -d -f openocd.cfg -c "init; halt; nrf51 mass_erase; program _build/nrf52832_xxaa_s132_patched.bin verify; reset; exit";`

I get this:

```
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 3 1 options.c:52 configuration_output_handler(): debug_level: 3
User : 4 1 options.c:52 configuration_output_handler(): 
Debug: 5 1 options.c:233 add_default_dirs(): bindir=/usr/bin
Debug: 6 1 options.c:234 add_default_dirs(): pkgdatadir=/usr/share/openocd
Debug: 7 1 options.c:235 add_default_dirs(): exepath=/usr/bin
Debug: 8 1 options.c:236 add_default_dirs(): bin2data=../share/openocd
Debug: 9 1 configuration.c:33 add_script_search_dir(): adding /home/xxxxx/.config/openocd
Debug: 10 1 configuration.c:33 add_script_search_dir(): adding /home/xxxxx/.openocd
Debug: 11 1 configuration.c:33 add_script_search_dir(): adding /usr/bin/../share/openocd/site
Debug: 12 1 configuration.c:33 add_script_search_dir(): adding /usr/bin/../share/openocd/scripts
Debug: 13 1 command.c:155 script_debug(): command - ocd_find openocd.cfg
Debug: 14 1 configuration.c:88 find_file(): found openocd.cfg
Debug: 15 1 command.c:155 script_debug(): command - ocd_find interface/stlink-v2.cfg
Debug: 16 1 configuration.c:88 find_file(): found /usr/bin/../share/openocd/scripts/interface/stlink-v2.cfg
Debug: 17 1 command.c:155 script_debug(): command - echo WARNING: interface/stlink-v2.cfg is deprecated, please switch to interface/stlink.cfg
User : 18 1 command.c:685 handle_echo(): WARNING: interface/stlink-v2.cfg is deprecated, please switch to interface/stlink.cfg
Debug: 19 1 command.c:155 script_debug(): command - ocd_find interface/stlink.cfg
Debug: 20 1 configuration.c:88 find_file(): found /usr/bin/../share/openocd/scripts/interface/stlink.cfg
Debug: 21 1 command.c:155 script_debug(): command - adapter driver hla
Debug: 22 1 command.c:155 script_debug(): command - hla_layout stlink
Debug: 23 1 hla_interface.c:230 hl_interface_handle_layout_command(): hl_interface_handle_layout_command
Debug: 24 1 command.c:155 script_debug(): command - hla_device_desc ST-LINK
Debug: 25 1 hla_interface.c:217 hl_interface_handle_device_desc_command(): hl_interface_handle_device_desc_command
Debug: 26 1 command.c:155 script_debug(): command - hla_vid_pid 0x0483 0x3744 0x0483 0x3748 0x0483 0x374b 0x0483 0x374d 0x0483 0x374e 0x0483 0x374f 0x0483 0x3752 0x0483 0x3753 0x0483 0x3754
Debug: 27 1 command.c:155 script_debug(): command - ocd_find target/nrf52.cfg
Debug: 28 1 configuration.c:88 find_file(): found /usr/bin/../share/openocd/scripts/target/nrf52.cfg
Debug: 29 1 command.c:155 script_debug(): command - ocd_find target/swj-dp.tcl
Debug: 30 1 configuration.c:88 find_file(): found /usr/bin/../share/openocd/scripts/target/swj-dp.tcl
Debug: 31 1 command.c:155 script_debug(): command - transport select
Info : 32 1 transport.c:265 jim_transport_select(): auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Debug: 33 1 hla_transport.c:191 hl_swd_transport_select(): hl_swd_transport_select
Debug: 34 2 command.c:155 script_debug(): command - transport select
Debug: 35 2 command.c:155 script_debug(): command - expr  [ string first "jtag" $_TRANSPORT ] != -1 
Debug: 36 2 command.c:155 script_debug(): command - transport select
Debug: 37 2 command.c:155 script_debug(): command - expr  [ string first "swd" $_TRANSPORT ] != -1 
Debug: 38 2 command.c:155 script_debug(): command - swd newdap nrf52 cpu -expected-id 0x2ba01477
Debug: 39 2 hla_tcl.c:101 jim_hl_newtap_cmd(): Creating New Tap, Chip: nrf52, Tap: cpu, Dotted: nrf52.cpu, 2 params
Debug: 40 2 hla_tcl.c:112 jim_hl_newtap_cmd(): Processing option: -expected-id
Debug: 41 2 core.c:1474 jtag_tap_init(): Created Tap: nrf52.cpu @ abs position 0, irlen 0, capture: 0x0 mask: 0x0
Debug: 42 2 command.c:155 script_debug(): command - dap create nrf52.dap -chain-position nrf52.cpu
Debug: 43 2 command.c:155 script_debug(): command - target create nrf52.cpu cortex_m -dap nrf52.dap
Info : 44 2 target.c:6205 target_create(): The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Debug: 45 2 hla_target.c:193 adapter_target_create(): adapter_target_create
Debug: 46 2 hla_target.c:163 adapter_init_arch_info(): adapter_init_arch_info
Debug: 47 2 command.c:289 register_command(): command 'rtt' is already registered
Debug: 48 2 command.c:289 register_command(): command 'tpiu' is already registered
Debug: 49 2 command.c:155 script_debug(): command - adapter speed 1000
Debug: 50 2 adapter.c:251 adapter_config_khz(): handle adapter khz
Debug: 51 2 adapter.c:215 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 52 2 adapter.c:215 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 53 2 command.c:155 script_debug(): command - nrf52.cpu configure -work-area-phys 0x20000000 -work-area-size 0x4000 -work-area-backup 0
Debug: 54 2 target.c:2199 target_free_all_working_areas_restore(): freeing all working areas
Debug: 55 2 target.c:2199 target_free_all_working_areas_restore(): freeing all working areas
Debug: 56 2 target.c:2199 target_free_all_working_areas_restore(): freeing all working areas
Debug: 57 2 command.c:155 script_debug(): command - transport select
Debug: 58 2 command.c:155 script_debug(): command - expr  [ string first "hla" $_TRANSPORT ] != -1 
Debug: 59 2 command.c:155 script_debug(): command - echo 
[... 87 lines cut ...]
Debug: 147 11 stlink_usb.c:1130 stlink_usb_error_check(): unknown/unexpected STLINK status code 0x5
Debug: 148 11 target.c:2633 target_read_u32(): address: 0xe000ed00 failed
Debug: 149 11 target.c:1843 target_call_event_callbacks(): target event 20 (examine-fail) for core nrf52.cpu
Warn : 150 11 target.c:802 target_examine(): target nrf52.cpu examination failed
Debug: 151 11 openocd.c:153 handle_init_command(): target examination failed
Debug: 152 11 command.c:155 script_debug(): command - flash init
Debug: 153 11 tcl.c:1375 handle_flash_init_command(): Initializing flash devices...
Debug: 154 11 command.c:155 script_debug(): command - nand init
Debug: 155 11 tcl.c:487 handle_nand_init_command(): Initializing NAND devices...
Debug: 156 11 command.c:155 script_debug(): command - pld init
Debug: 157 11 pld.c:194 handle_pld_init_command(): Initializing PLDs...
Debug: 158 11 command.c:155 script_debug(): command - tpiu init
Info : 159 11 gdb_server.c:3791 gdb_target_start(): starting gdb server for nrf52.cpu on 3333
Info : 160 11 server.c:297 add_service(): Listening on port 3333 for gdb connections
Debug: 161 11 command.c:155 script_debug(): command - halt
Debug: 162 11 target.c:3305 handle_halt_command(): -
Error: 163 11 target.c:590 target_halt(): Target not examined yet
Debug: 164 11 command.c:544 run_command(): Command 'halt' failed with error code -4
User : 165 11 command.c:608 command_run_line(): 
Debug: 166 11 target.c:2199 target_free_all_working_areas_restore(): freeing all working areas
Debug: 167 11 hla_interface.c:119 hl_interface_quit(): hl_interface_quit
Debug: 168 12 stlink_usb.c:1659 stlink_usb_exit_mode(): MODE: 0x02
```

Any idea what I am doing wrong?

pix, 2024-11-13:
This may be a duplicate of https://github.com/pix/heystack-nrf5x/issues/4

Stock ST-Link V2 are unable to flash nrf52 when AP-PROTECT is enabled:

> Debug: 61 2 command.c:155 script_debug(): command - echo nRF52 device has a CTRL-AP dedicated to recover the device from AP lock.
> User : 62 2 command.c:685 handle_echo(): nRF52 device has a CTRL-AP [...]

tobox, 2024-11-14:
You nailed it.

I flashed jlink to my ST-Link V2 and was able to remove the protection. It works now!

Should I use BMP instead of JLink?

closed 2024-11-14 by tobox

tobox, 2024-11-14:
Instruction on how to flash jlink on st-link flasher (Windows only):

[Flash jlink firmware to st-link hardware](https://www.segger.com/products/debug-probes/j-link/models/other-j-links/st-link-on-board/)

You may need to patch some bytes in the executable file...

[Patch bytes in STLinkReflash](https://gist.github.com/jamesy0ung/919ef51fea4631e9dfe0bd26dc85c8f0)

Then I continued on linux. To remove the protection, I found this snippet that worked:

`openocd -c 'adapter driver jlink; transport select swd; source [find target/nrf52.cfg]' -c init -c 'nrf52.dap apreg 1 0xc' -c 'nrf52.dap apreg 1 0x4 1'`

Then I used openocd to finally flash the file:

`openocd -c "adapter driver jlink;transport select swd; source [find target/nrf52.cfg]; init; halt; nrf51 mass_erase; program _build/nrf52832_xxaa_s132_patched.bin verify; reset; exit"`

pix, 2024-11-14:
You don’t necessarily need to switch to BMP, but it could be beneficial. BMP firmware allows you to use the monitor feature, which can be useful if you need it. I remember attempting to flash the JLink firmware on my V2 clones without success, but BMP worked seamlessly right out of the box. Additionally, BMP is fully open-source and legally licensed.