pix/heystack-nrf5x #10: nRF51822 seems to not work after flashing

KlausMu, 2024-11-17:
I went through the README to flash a nRF51822. To my understanding, everything worked as expected.
But after flashing, I can't see the tag, neither in "Android nRF Connect", nor can the tag being added to "Find My" in an iPhone.

The tag is from AliExpress (HolyIoT) and looks like this:
[image] 
[image]
[image]

The command for flashing was
```
make stflash-nrf51822_xxac-patched ADV_KEYS_FILE=/home/klaus/projects/airtag/heystack-nrf5x/tools/output-nrf51822-2/nrf51822-2_keyfile ADVERTISING_INTERVAL=2000 KEY_ROTATION_INTERVAL=3600*12 MAX_KEYS=50 HAS_DEBUG=1
```

Log while flashing was:
```
merging application and softdevice... : nrf51822_xxac
mergehex -m _build/nrf51822_xxac.hex ../../nrf-sdk/nRF5_SDK_12.3.0_d7731ad/components/softdevice/s130/hex/s130_nrf51_2.0.1_softdevice.hex -o _build/nrf51822_xxac_s130.hex
Parsing input files.
Merging file "nrf51822_xxac.hex" into output.
Merging file "s130_nrf51_2.0.1_softdevice.hex" into output.
Storing merged file.
hex2bin.py _build/nrf51822_xxac_s130.hex _build/nrf51822_xxac_s130.bin
Patching nrf51822_xxac
cp _build/nrf51822_xxac_s130.bin _build/nrf51822_xxac_s130_patched.bin
xxd -p -c 100000 /home/klaus/projects/airtag/heystack-nrf5x/tools/output-nrf51822-2/nrf51822-2_keyfile | xxd -r -p | dd of=_build/nrf51822_xxac_s130_patched.bin skip=1 bs=1 seek=116704 conv=notrunc
1400+0 records in
1400+0 records out
1400 bytes (1.4 kB, 1.4 KiB) copied, 0.00267036 s, 524 kB/s
xxd -p -c 100000 _build/nrf51822_xxac_s130_patched.bin | grep -q 454e444f464b455953454e444f464b455953454e444f464b45595321 || (echo "The key was not patched correctly!"; exit 1)
"/home/klaus/projects/airtag/heystack-nrf5x/nrf-sdk/gcc-arm-none-eabi-6-2017-q2-update/bin/arm-none-eabi-objcopy" -I binary -O elf32-littlearm -B arm _build/nrf51822_xxac_s130_patched.bin _build/nrf51822_xxac_s130_patched.elf
flashing application and softdevice...␍
openocd -f openocd.cfg -c "init; halt; nrf51 mass_erase; program _build/nrf51822_xxac_s130_patched.bin verify; reset; exit";
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V2J37S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.098182
Info : [nrf51.cpu] Cortex-M0 r0p0 processor detected
Info : [nrf51.cpu] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for nrf51.cpu on 3333
Info : Listening on port 3333 for gdb connections
[nrf51.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x00011434 msp: 0x20003fd8
Info : nRF51822-QFAA(build code: H0) 256kB Flash, 16kB RAM
Info : Mass erase completed.
Info : A reset or power cycle is required if the flash was protected before.
[nrf51.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
Warn : Adding extra erase range, 0x0001ce20 .. 0x0001cfff
** Programming Finished **
** Verify Started **
** Verified OK **
```
Log from strtt is:
```
./strtt -v 4
Debug: 1 0 stlink.c:4163 stlink_open(): stlink_open
Debug: 2 0 stlink.c:4177 stlink_open(): transport: 4 vid: 0x0483 pid: 0x3748 serial: not provided
Debug: 3 0 stlink.c:4177 stlink_open(): transport: 4 vid: 0x0483 pid: 0x374b serial: not provided
Debug: 4 0 stlink.c:4177 stlink_open(): transport: 4 vid: 0x0483 pid: 0x3752 serial: not provided
Debug: 5 0 stlink.c:4177 stlink_open(): transport: 4 vid: 0x0483 pid: 0x374d serial: not provided
Debug: 6 0 stlink.c:4177 stlink_open(): transport: 4 vid: 0x0483 pid: 0x374e serial: not provided
Debug: 7 0 stlink.c:4177 stlink_open(): transport: 4 vid: 0x0483 pid: 0x374f serial: not provided
Debug: 8 0 stlink.c:4177 stlink_open(): transport: 4 vid: 0x0483 pid: 0x3753 serial: not provided
Debug: 9 0 stlink.c:4177 stlink_open(): transport: 4 vid: 0x0483 pid: 0x3754 serial: not provided
Info : 10 3 stlink.c:1507 stlink_usb_version(): STLINK V2J37S7 (API v2) VID:PID 0483:3748
Debug: 11 4 stlink.c:1735 stlink_usb_exit_mode(): MODE: 0x00
Info : 12 58 stlink.c:1543 stlink_usb_check_voltage(): Target voltage: 3.099327
Debug: 13 58 stlink.c:1808 stlink_usb_init_mode(): MODE: 0x01
Debug: 14 58 stlink.c:3479 stlink_dump_speed_map(): Supported clock speeds are:
Debug: 15 58 stlink.c:3482 stlink_dump_speed_map(): 4000 kHz
Debug: 16 58 stlink.c:3482 stlink_dump_speed_map(): 1800 kHz
Debug: 17 58 stlink.c:3482 stlink_dump_speed_map(): 1200 kHz
Debug: 18 58 stlink.c:3482 stlink_dump_speed_map(): 950 kHz
Debug: 19 58 stlink.c:3482 stlink_dump_speed_map(): 480 kHz
Debug: 20 58 stlink.c:3482 stlink_dump_speed_map(): 240 kHz
Debug: 21 58 stlink.c:3482 stlink_dump_speed_map(): 125 kHz
Debug: 22 58 stlink.c:3482 stlink_dump_speed_map(): 100 kHz
Debug: 23 58 stlink.c:3482 stlink_dump_speed_map(): 50 kHz
Debug: 24 58 stlink.c:3482 stlink_dump_speed_map(): 25 kHz
Debug: 25 58 stlink.c:3482 stlink_dump_speed_map(): 15 kHz
Debug: 26 58 stlink.c:3482 stlink_dump_speed_map(): 5 kHz
Debug: 27 62 stlink.c:1876 stlink_usb_init_mode(): MODE: 0x02
Debug: 28 62 stlink.c:4356 stlink_usb_init_access_port(): init ap_num = 0
Debug: 29 63 stlink.c:4503 stlink_usb_open_ap(): AP 0 enabled
Debug: 30 65 stlink.c:4261 stlink_open(): Using TAR autoincrement: 1024
Debug: 31 186 strtt.cpp:160 findRtt(): RTT addr = 0x200029e8
Debug: 32 186 strtt.cpp:186 findRtt(): Max number of buffers UP: 2 and DOWN: 2
Info : 33 188 strtt.cpp:215 getRttDesc(): 0. Channel name: Terminal     size: 512       mode: 0
Info : 34 188 strtt.cpp:221 getRttDesc(): 1. Channel name: -------      size: 0 mode: 0
Info : 35 189 strtt.cpp:215 getRttDesc(): 2. Channel name: Terminal     size: 16        mode: 0
Info : 36 189 strtt.cpp:221 getRttDesc(): 3. Channel name: -------      size: 0 mode: 0
Debug: 37 199 strtt.cpp:326 readRtt(): Chanel: 0 readed: 440
:INFO:[KEYS] Last filled index: 48
:INFO:[TIMING] Full key rotation interval: 2073600 seconds (576.00 hours)
:INFO:[TIMING] Rotation per Day: 0.04
:INFO:Starting advertising
:INFO:ble_set_mac_address: eb:fe:8c:1b:xx:xx
:INFO:ble_set_max_tx_power: 8 dBm failed
:INFO:ble_set_max_tx_power: 7 dBm failed
:INFO:ble_set_max_tx_power: 6 dBm failed
:INFO:ble_set_max_tx_power: 5 dBm failed
:INFO:ble_set_max_tx_power: 4 dBm
:INFO:Rotating key: 46
```

pix, 2024-11-19:
I'll take a look, It seems https://github.com/pix/heystack-nrf5x/issues/7 has the same problem.

dirkpauli, 2024-12-31:
not 100% sure if this is the same problem I have - but it feels like. The only difference is: I'm using HolyIOT's 21011 and so flashing the nRF52810 version of the firmware.

dirkpauli, 2025-01-02:
just fyi: I managed the project to get working by resetting the git to commit 1efcb92. A Firmware with that built works fine on my nRF52810.

KlausMu, 2025-01-04:
Thanks for the hint. I tried commit https://github.com/pix/heystack-nrf5x/commit/1efcb926d963cb63606f03361c7eb5f4178b8e65
With an nRF51822 and this command `make stflash-nrf51822_xxac-patched ADV_KEYS_FILE=./50_NRF_keyfile` , flashing gives an error with 
```
Parsing input files.
Could not find file "nrf51822_xxac.hex"
ERROR: One of the files to merge cannot be found.
make: *** [../../Makefile.common:62: merge_nrf51822_xxac] Error 51
```
So I tried to compile with master and to flash file `nrf51822_xxac.hex`with commit https://github.com/pix/heystack-nrf5x/commit/1efcb926d963cb63606f03361c7eb5f4178b8e65
Flashing was successfull, but device still cannot be seen by the iPhone.

skrobul, 2025-01-05:
This is probably too late, but I also had issues when using this exact board (YJ-15044). Eventually I got it to work though, with current master (c5cb169eb90886bd99dcd0c2b1afef40d9ee830c).
Here are some notes from the issues I experienced, maybe they will help someone

#### Android nRFConnect does not display FindMy tags

Turns out my issue was with the way I was debugging  the "nRF Connect" app on Android displays tens of different bluetooth devices, but by default it filters out "Microsoft, Samsung, Google and Apple" devices... After adjusting filter on the dropdown and unchecking "Apple" i started seeing the FindMy tag.

#### Flashing on RaspberryPI
- OpenOCD from the raspbian repo did not work. Followed instructions [from this article](https://sysadmin-central.com/2020/12/05/connecting-raspberry-pi-4-to-nrf51822-bluetooth-chip-using-openocd/) to compile and setup the RaspberryPi profile, but that also did help immediately. 
- This again turned out to be a user problem, since  I swapped the pins. It was hard to find which ones to use in the first place - they are configured [here](https://github.com/arduino/OpenOCD/blob/c404ff5d3a2ec568daa106455845dd403b08dab4/tcl/interface/raspberrypi2-native.cfg#L29) and with default setup it's PIN `GPIO25` for `SWCLCK` and PIN `GPIO24` for `SWDIO`. This translates to physical pins 22 and 18 respectively on Rpi4B. 
- in `openocd.conf` I needed to switch transport from `hla_swd` to `swd` as `hla_swd` seems to be stlink specific.
- In `openocd.conf` I needed to adjust `set WORKAREASIZE 0x4000` to `set WORKAREASIZE 0` in order to force "slow memory writes".

supaeasy, 2025-03-03:
@skrobul @KlausMu 
You might want to have a look [here](https://github.com/acalatrava/openhaystack-firmware/blob/main/apps/openhaystack-alternative/iBeacon%20StepByStep.md). Some time ago I wrote a Step-by-Step for future reference for myself since it was such a struggle to get there. It is about the exact same tag.

supaeasy, 2025-03-03:
also @KlausMu It looks like your red wire (and maybe also the black) is soldered to the wrong place. Not entirely sure though, it is hard to see.

KlausMu, 2025-03-09:
Thanks for your hint, @supaeasy I flashed a patched precompiled firmware from https://github.com/acalatrava/openhaystack-firmware.
Unfortunately with the same result: my iPhone 7 simply doesn't see the nRF51822.

The only thing I am not certain about is the advertisement key. Since I don't have a Mac, I can't use https://github.com/seemoo-lab/openhaystack to generate one.
So I took one generated with the scripts provided here on this repo https://github.com/pix/heystack-nrf5x
This is something like `D/Q06GEV/wkdJDhfzr645tdgEWdRpv5Lqa0Fww==` (not my real key)
So the complete command for patching was `NRF_MODEL=nrf51 ADV_KEY_BASE64=D/Q06GEV/wkdJDhfzr645tdgEWdRpv5Lqa0Fww== make patch`
And I had to change `base64 -D` to `base64 -d` in the Makefile.

Wires were correct, there are several ways to connect to GND.

KlausMu, 2025-03-09:
@supaeasy I now also tested with an iPhone 13. But the tag cannot be seen by the iPhone.
Am I missing something? Do I have to put the tag into a pairing mode? But the nRF51822 does not have any button.
The Apple airtag or any other third party airtag always has a pairing button. But to my understanding, the heystack firmware automatically announces. Is this correct?

skrobul, 2025-03-09:
@KlausMu are you definitely using a Heystack application to add the tag?- it will never be seen by the iPhone's native application (FindMy or whatever it is called). Just checking, because initially I was also under an impression that you could pair it just as you'd pair genuine tag, but that simply isn't the case.

In regards to not having a mac - have a look at https://github.com/dchristl/macless-haystack - iirc there is a python script there to generate own keys that definitely work (tested on 4 tags).

KlausMu, 2025-03-09:
I think I already did everything which is mentioned on https://github.com/dchristl/macless-haystack. But I will check again.

If it is not possible to pair a custom firmware tag with the iPhone app, how is it done? You have to pair in it some way to your Apple ID, right?

skrobul, 2025-03-09:
>  You have to pair in it some way to your Apple ID, right?

No, you don't. Apple does not know nor tracks who does particular tag belong to. Your Apple ID is only used to retrieve the location reports. You can retrieve history of location for any tag on earth, but only their owners have access to the private keys used to decrypt those location reports.

Now, equivalent of the "registration" process is just uploading your private key to the Heystack app (or findme.py or similar apps).

If that does not make sense and you are interested in learning more, reading [the original "Who Can Find My Devices? Security and Privacy of Apple’s Crowd-Sourced Bluetooth Location Tracking System"](https://www.petsymposium.org/2021/files/papers/issue3/popets-2021-0045.pdf) helped me a ton to understand how this works.
Alternatively, [the MAKE magazine article](https://makezine.com/projects/find-my-diy-airtag-tracker) was also good and less academic.

supaeasy, 2025-03-09:
@KlausMu 
Just to clarify: this is NOT about creating your own AirTag. Neither will the Original FindMy app work nor will any of airtag functions. 

This is about using the FindMy **network** with custom tags and a custom app.

To be clear it actually is better than the original AirTags with the same reliability.

KlausMu, 2025-03-15:
@skrobul @supaeasy thanks for pointing this out. I indeed misunderstood this. I cannot register such a tag in the iPhone FindMy app. But I was able to retrieve location with the scripts provided [here](https://github.com/biemster/FindMy.git). I had to wait some time until location was available, but finally it worked.

closed 2025-03-15 by KlausMu