pix/heystack-nrf5x #12: Cannot find "gcc-arm-none-eabi-6-2017-q2-update/arm-none-eabi-gcc"

mm28ajos, 2024-12-14:
Hey there,

I am trying to flash a HolyIoT 21011 with a nRF52810 chip. However, I could not resolve the following issue, see blow. As I understand the issue, there is no `arm-none-eabi-gcc` folder but only a folder `arm-none-eabi`  in the `gcc-arm-none-eabi-6-2017-q2-update` SDK? What confuses me is, that the `GNU_PREFIX` is already `arm-none-eabi` (so no postfix "-gcc" there).

Could you give me a hint what to check / change?

Thanks!

**Error message** 
```
user@host:~/Documents/Git/heystack-nrf5x/nrf52810/armgcc$ make help-msg
make: /home/user/Documents/Git/heystack-nrf5x/nrf-sdk/gcc-arm-none-eabi-6-2017-q2-update/arm-none-eabi-gcc: No such file or directory
Cannot find: '/home/user/Documents/Git/heystack-nrf5x/nrf-sdk/gcc-arm-none-eabi-6-2017-q2-update/arm-none-eabi-gcc'.
Please set values in: "/home/user/Documents/Git/heystack-nrf5x/nrf-sdk/nRF5_SDK_15.3.0_59ac345/components/toolchain/gcc/Makefile.posix"
according to the actual configuration of your system.
../../nrf-sdk/nRF5_SDK_15.3.0_59ac345/components/toolchain/gcc/Makefile.common:129: *** Cannot continue.  Stop.
```

**My Makefile.posix** from /home/user/Documents/Git/heystack-nrf5x/nrf-sdk/nRF5_SDK_15.3.0_59ac345/components/toolchain/gcc/Makefile.posix

```
GNU_INSTALL_ROOT ?= /home/user/Documents/Git/heystack-nrf5x/nrf-sdk/gcc-arm-none-eabi-6-2017-q2-update/
GNU_VERSION ?= 7.3.1
GNU_PREFIX ?= arm-none-eabi
```

Content of ~/Documents/Git/heystack-nrf5x/nrf-sdk/gcc-arm-none-eabi-6-2017-q2-update
```
drwxr-xr-x user user 4 KB Tue Jun 27 10:16:17 2017   ./            
drwxrwxr-x user user 4 KB Thu Dec 12 22:51:28 2024   ../           
drwxr-xr-x user user 4 KB Wed Jun 21 11:17:21 2017   arm-none-eabi/
drwxr-xr-x user user 4 KB Wed Jun 21 11:20:21 2017   bin/          
drwxr-xr-x user user 4 KB Wed Jun 21 11:20:21 2017   lib/          
drwxr-xr-x user user 4 KB Wed Jun 21 11:02:54 2017   share/ 
```

sonman, 2024-12-16:
Hi, 

what worked for me:
`
cd /fw/nrf-sdk/gcc-arm-none-eabi-6-2017-q2-update/
ln -s bin/arm-none-eabi-gcc arm-none-eabi-gcc
ln -s bin/arm-none-eabi-objcopy arm-none-eabi-objcopy
ln -s bin/arm-none-eabi-size arm-none-eabi-size
`

mm28ajos, 2024-12-18:
Thanks a lot! Die the trick.