pix/heystack-nrf5x #13: NRF852810 Flashing does not start

mic-bec, 2024-12-20:
This is my first attempt to flash with a ST-Link V2 and I have a problem starting the flash process on an nrf52810 tag. 
After the command to flash, the process stops at this point.

cp _build/nrf52810_xxaa_s112.bin _build/nrf52810_xxaa_s112_patched.bin
xxd -p -c 100000  | xxd -r -p | dd of=_build/nrf52810_xxaa_s112_patched.bin skip=1 bs=1 seek=109128 conv=notrunc

Any ideas what I could look for? I think I have connected all 4 lines correctly.  Thanks.

lovelyelfpop, 2024-12-23:
Any error logs?

mic-bec, 2024-12-23:
Nothing more on the screen. Is there a logfile I could look for?

pix, 2024-12-30:
Hi. Sorry, I have some personal issues preventing me from working on this, but I think the ADV_KEYS_FILE arg is missing, so it waits on STDIN for key material (not ideal)

mic-bec, 2024-12-31:
That's it. Thank you very very much.
I didn't forget the ADV_KEYS_FILE, but I made a syntax failure.

pix, 2025-01-08:
No problems

closed 2025-01-08 by pix