pix/heystack-nrf5x #15: Unable to merge firmware

jjcszxh, 2025-01-05:
Error log：
../../nrf-sdk/nRF5_SDK_12.3.0_d7731ad/components/toolchain/gcc/Makefile.common:25: *** Cannot continue.  Stop.
PS C:\Users\Administrator\Downloads\heystack-nrf5x\nrf51822\armgcc> make stflash-nrf51822_xxac-patched GNU_INSTALL_ROOT=../../nrf-sdk/gcc-arm-none-eabi-6-2017-q2-update-win32/ ADV_KEYS_FILE=../../tools/output-SmallTag1/SmallTag1_keyfile HAS_DEBUG=1
mkdir _build
Compiling file: nrf_log_backend_serial.c
Compiling file: nrf_log_frontend.c
Compiling file: app_button.c
Compiling file: app_error.c
Compiling file: app_error_weak.c
Compiling file: app_timer.c
Compiling file: app_util_platform.c
Compiling file: hardfault_implementation.c
Compiling file: nrf_assert.c
Compiling file: sdk_errors.c
Compiling file: boards.c
Compiling file: nrf_drv_clock.c
Compiling file: nrf_drv_common.c
Compiling file: nrf_drv_gpiote.c
Compiling file: nrf_drv_uart.c
Compiling file: bsp.c
Compiling file: bsp_nfc.c
Compiling file: main.c
In file included from ../../main.c:54:0:
../../main.h:43:9: note: #pragma message: RTC_FREQUENCY: (32768 / (31 + 1))
 #pragma message("RTC_FREQUENCY: " _STRINGIFY(COMPUTED_RTC_FREQUENCY))
         ^~~~~~~
../../main.h:44:9: note: #pragma message: MAX_TIMER_INTERVAL_SECONDS: (0xFFFFFF / (32768 / (31 + 1)))
 #pragma message("MAX_TIMER_INTERVAL_SECONDS: " _STRINGIFY(COMPUTED_MAX_TIMER_INTERVAL))
         ^~~~~~~
Compiling file: ble_stack.c
Compiling file: RTT_Syscalls_GCC.c
Compiling file: SEGGER_RTT.c
Compiling file: SEGGER_RTT_printf.c
Compiling file: ble_advdata.c
Compiling file: ble_conn_params.c
Compiling file: ble_srv_common.c
Assembling file: gcc_startup_nrf51.S
../../nrf-sdk/nRF5_SDK_12.3.0_d7731ad/components/toolchain/gcc/Makefile.common:145: warning: pattern recipe did not update peer target '_build/nrf51822_xxac_gcc_startup_nrf51.s.o.o'.
Compiling file: system_nrf51.c
Compiling file: softdevice_handler.c
Linking target: _build/nrf51822_xxac.out

   text    data     bss     dec     hex filename
  14180     528    2156   16864    41e0 _build/nrf51822_xxac.out

Preparing: _build/nrf51822_xxac.hex
Preparing: _build/nrf51822_xxac.bin
merging application and softdevice... : nrf51822_xxac
mergehex -m _build/nrf51822_xxac.hex ../../nrf-sdk/nRF5_SDK_12.3.0_d7731ad/components/softdevice/s130/hex/s130_nrf51_2.0.1_softdevice.hex -o _build/nrf51822_xxac_s130.hex
Parsing input hex files.
Merging files.
Storing merged file.
hex2bin.py _build/nrf51822_xxac_s130.hex _build/nrf51822_xxac_s130.bin
process_begin: CreateProcess(NULL, hex2bin.py _build/nrf51822_xxac_s130.hex _build/nrf51822_xxac_s130.bin, ...) failed.
make (e=2):
make: *** [../../Makefile.common:84: bin_nrf51822_xxac] Error 2

5Breeze, 2025-02-20:
I meet the same question.......Have you found the way to solve it?:)`root@HomeService:/home/FindMy/heystack-nrf5x-master# make all
dirname: missing operand
Try 'dirname --help' for more information.
make -C nrf51822/armgcc GNU_INSTALL_ROOT=/home/FindMy/heystack-nrf5x-master/nrf-sdk/gcc-arm-none-eabi-6-2017-q2-update/ MAX_KEYS=500 HAS_DEBUG=0 HAS_BATTERY=0 KEY_ROTATION_INTERVAL=3600 ADVERTISING_INTERVAL=1000 RANDOM_ROTATE_KEYS=1 nrf51822_xxac bin_nrf51822_xxac
make[1]: Entering directory '/home/FindMy/heystack-nrf5x-master/nrf51822/armgcc'
make[1]: Nothing to be done for 'nrf51822_xxac'.
merging application and softdevice... : nrf51822_xxac
mergehex -m _build/nrf51822_xxac.hex ../../nrf-sdk/nRF5_SDK_12.3.0_d7731ad/components/softdevice/s130/hex/s130_nrf51_2.0.1_softdevice.hex -o _build/nrf51822_xxac_s130.hex
make[1]: mergehex: No such file or directory
make[1]: *** [../../Makefile.common:72: merge_nrf51822_xxac] Error 127
make[1]: Leaving directory '/home/FindMy/heystack-nrf5x-master/nrf51822/armgcc'
make: *** [Makefile:66: nrf51822_xxac] Error 2`

5Breeze, 2025-02-20:
OK, I find the answer, you need to download the deb (or other system) packs from `https://www.nordicsemi.com/Products/Development-tools/nrf-command-line-tools/download`, after that , you need install the mergehex tool.