pix/heystack-nrf5x #17: Cannot customise the Makefile variables

sachsflo, 2025-02-17:
First of all: I have very little experience with make. Accordingly, the error may also be mine.

I am currently working with nrf52810.

Make-Command (some values are for testing purposes):
make stflash-nrf52810_xxaa-patched ADV_KEYS_FILE=/root/FindMy/heystack-nrf5x/tools/output/D3GALA_keyfile KEY_ROTATION_INTERVAL=360 MAX_KEYS=10 ADVERTISING_INTERVAL=20000 HAS_BATTERY=1

or (I am flashing on another device)
make nrf52810_xxaa-patched ADV_KEYS_FILE=/root/FindMy/heystack-nrf5x/tools/output/D3GALA_keyfile KEY_ROTATION_INTERVAL=3600/10 MAX_KEYS=10 ADVERTISING_INTERVAL=20000 HAS_BATTERY=1

The whole thing seems to run without errors. The key file is also patched successfully.
However, the other settings are not applied, the beacon continues to transmit with ADVERTISING_INTERVAL=1000 and KEY_ROTATION_INTERVAL 1 hour.

I then tried to change the values directly in the makefile, also without success.
Even a make clean (both in /heystack-nrf5x and in /nrf52810/armgcc) does not bring any change.

Any recommendations?

magnets110, 2025-02-21:
same for me with the HOLYIOT 21014 nrf52810

power pulses every 1 second

magnets110, 2025-02-21:
put your settings in Makefile in the root directory then run 'make all' from there

navigate to the nrf52810/armgcc subdir and run 'make stflash-nrf52810_xxaa-patched ADV_KEYS_FILE=./50_NRF_keyfile'