pix/heystack-nrf5x #23: YJ-17024 target nrf52.cpu examination failed

sonman, 2025-03-31:
Hi, when running `openocd -f nrf52832/armgcc/openocd.cfg -c "init;halt;"` I get following output

```
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD

nRF52 device has a CTRL-AP dedicated to recover the device from AP lock.
A high level adapter (like a ST-Link) you are currently using cannot access
the CTRL-AP so 'nrf52_recover' command will not work.
Do not enable UICR APPROTECT.

Info : clock speed 1000 kHz
Info : STLINK V2J37S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.091907
Warn : target nrf52.cpu examination failed
Info : starting gdb server for nrf52.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet
```

Not sure what causes it.

pix, 2025-04-07:
The error is right here:

> nRF52 device has a CTRL-AP dedicated to recover the device from AP lock.
> A high level adapter (like a ST-Link) you are currently using cannot access
> the CTRL-AP so 'nrf52_recover' command will not work.

Your ST-Link can't deactivate CTRL-AP. You neet to flash your ST-Link with a BlackMagicProbe firmware, or use another programmer.

sonman, 2025-04-07:
Since I got the same "error" about the CTRL-AP on a YJ-21014 but there the flashing works I assumed that also on a YJ-17024 the AP is not locked. But I may was wrong.
I going try another programmer. Thanks.

closed 2025-07-01 by sonman

sonman, 2025-07-01:
Just another note:
After downgrading my stlink programming worked BUT only one time. Then the STLink Version strangely shows as V15J63S63 and flashing fails with "init mode failed". Reflashing again works for exact one flashing procedure. Strange...

sonman, 2025-07-01:
Updating to latest firmware V2J46S7 seems not to have this issue (but have not tested to unlock protection with this version)