pix/heystack-nrf5x #25: nRF51822 no longer being recognized - no more reports from FindMy network

KlausMu, 2025-06-04:
I flashed 5 nRF51822.
[image]
All 5 of them have been working well for about 8 weeks.

But suddenly, at the same point in time, 4 of them stopped being recognized. I cannot longer retrieve reports for them from the FindMy network. Only for 1 device I still get reports.

But the 4 other devices still seem to send out messages:

- I can see them in "nRF Connect" in Android
- I can see debug messages with strtt
- Battery is almost full

I am using

- ADVERTISING_INTERVAL=2000
- KEY_ROTATION_INTERVAL=3600*12
- MAX_KEYS=50

I also tried with other values, e.g. advertising interval 1000 or much shorter rotation interval. I can see in "nRF Connect" and strtt that all options are correctly recognized.

So the devices seem to work completely correct. But I do not get any longer reports for 4 of them. Only 1 is still working.

Any idea what I could do?

dakhnod, 2025-09-17:
Dito. I assume that Apple now ignores advertisements it has already seen before?

dakhnod, 2025-09-17:
@sonman what makes you think so?

sonman, 2025-09-17:
See here https://github.com/biemster/FindMy/issues/82

KlausMu, 2025-09-27:
I can confirm that the nRF51822 is indeed still being recognized. But Apple limits the requested keys. If I request less keys, I can find the tags.

dakhnod, 2025-09-27:
@KlausMu how many keys are you successfully requesting?

KlausMu, 2025-09-27:
Don't know the exact limit. People at https://github.com/biemster/FindMy/issues/82 report a limit of 20 to 40
My 5 tags have 50 keys each. If I request reports for all tags (250 keys), I only get reports for one tag. If I request each tag separately (with only 50 keys), I get reports for all of them.

lovelyelfpop, 2025-09-29:
https://github.com/biemster/FindMy/issues/82#issuecomment-2915238736
https://github.com/dchristl/macless-haystack/issues/194#issuecomment-3240641925